chipsalliance/aib-phy-hardware #4: synthesis of the aib

liuzixuan1212, 2020-02-04:
Hi

I want to use this interface in my new project. Does this RTL code have been synthesis before. If it has been synthesis, can you tell me about your synthesis environment and how it is synthesis.

Thanks Zixuan

rahulrvm82, 2020-02-12:
> Hi
> 
> I want to use this interface in my new project. Does this RTL code have been synthesis before. If it has been synthesis, can you tell me about your synthesis environment and how it is synthesis.
> 
> Thanks Zixuan

Hello Zixuan,

I am also planning to use the same for my project.
I want to explore the constraint for this design if implemented .  
If you have any update please ping me.

Thanks 

Rahul

dkehlet, 2020-02-19:
aib-phy-hardware/rev2/rtl/v2_master and aib-phy-hardware/rev2/rtl/v2_common have been synthesized before, with the important exception: v2_common/ana are macrocells that were built using a custom methodology.
We don't have synthesis support yet but hope to add this in the future.

Dave

rahulrvm82, 2020-02-27:
Thanks For your response.

Can we have macro.vh file in data base?

This Error has stopped me: Invalid Macro 'AIBADAPTWRAPTCB_SCAN_CHAINS_NUM

Thanks 
Rahul

dkehlet, 2020-02-27:
Hi Rahul, please look in 
    aib-phy-hardware/rev2/rtl/v1_master/c3dfx/rtl/defines/c3dfx.vh
if you are using the rev1 files, or
    aib-phy-hardware/rev2/rtl/v2_common/c3dfx/rtl/defines/c3dfx.vh
if you are using the rev2 files.

Dave

$ ls
docs/  LICENSE*  README.txt*  rev1/  rev2/
$ grep -r "define AIBADAPTWRAPTCB_SCAN_CHAINS_NUM" .
./rev2/rtl/v1_master/c3dfx/rtl/defines/c3dfx.vh:`define AIBADAPTWRAPTCB_SCAN_CHAINS_NUM 17
./rev2/rtl/v2_common/c3dfx/rtl/defines/c3dfx.vh:`define AIBADAPTWRAPTCB_SCAN_CHAINS_NUM 17
$

xinyang2k, 2020-03-04:
if you look under rev1/rtl, there is a rtl model that have dummy dcc and dll module which is not doing anything relates to that. You can reference from there. You can use that model directly.

closed 2020-03-04 by xinyang2k

dkehlet, 2020-04-22:
Please see the README.md in rev2/constraints, just added.