chipsalliance/aib-phy-hardware #31: Ioring/Io_buffer Race Condition

Gatorman11, 2020-05-20:
Hello,

There is a race condition in the AIB Spec IP in simulation, specifically in the io_buffer/ioring.  I have attached a document outlining the scenarios, error situations, and solution that we found while working with the AIB Spec IP.

Can you please investigate this and provide a solution?

Thanks,
Eric

[Spec IP Race Condition.pdf](https://github.com/chipsalliance/aib-phy-hardware/files/4659849/Spec.IP.Race.Condition.pdf)

dkehlet, 2020-05-21:
Hi Eric, can you tell us which simulation case you are running?  
Also which simulator you are using?
And, are you looking at the aib_ioring.v on the Slave side of the simulation?

Gatorman11, 2020-05-21:
Hi Dave,

Thanks for the quick response. 

1)  The "working" simulation is just a straight copy from the ndsimslv rev1 directory.  We have the v2 spec IP channel 0 directly connected to a c3aibadapt instance.  The "broken" simulation uses the same scenario, but the c3aibadapt is instantiated within other IP, and an EMIB is used to connect the c3aibadapt instances with the spec IP channels.  Functionally the simulations do the same thing, but the addition of extra logic in the simulation seems to expose the race condition in the receiver.

2) We are using VCS-MX (P-2019.06-1)

3) Yes, the spec IP is the slave and the c3aibadapt is the master.  The race condition is seen in the slave side spec IP RTL.

Please let me know if you need anything else.

Thanks,
Eric

dkehlet, 2020-05-21:
Hi Eric, we acknowledge there is an issue (clock and data changing simultaneously) but we don't have consensus on the solution. In my view the DLL model output should have some delay applied (you can see in dll.sv it's actually there) but the DLL output is not sent back to aib_ioring.v for data sampling.

Your modification with the "#1" delay looks to me as a good workaround. If you were to submit a pull request with this change I would merge it.

For others' sake, please note that this problem occurs in an AIB model, not in the actual c3aibadapt RTL design.

dkehlet, 2021-02-09:
Will close in a week is there is no further discussion.

xinyang2k, 2021-04-05:
This issue has been fixe in v2.0/rev1. The delayed dll output go to aib_ioring.sv.

closed 2021-04-05 by xinyang2k