chipsalliance/aib-phy-hardware #34: Question about the rev1/ndsimslv/ testbench

liuzixuan1212, 2020-06-14:
Hi,

I have some question about the testbench in rev1/ndsimslv/ folder. 
The first question is how to modify the testbench to let the fpga aib module(slave) sends data and the master aib module receive data and loopback to the fpga aib module side? 
The second question is how to cancel the alignment mode. whether it need to change the configuration bits both in the master aib side and slave(fpga) aib side, and in that case how to change the configuration bits.  

Sincerely
Zixuan Liu

dkehlet, 2020-07-20:
1st question: This can be done; it just needs to be developed.  Can you be more specific please?
2nd question: You should review all the examples.  Also refer to the open_source_aib_csr.xlsx in the docs folder.

dkehlet, 2021-02-09:
Will close in a week if there is no further discussion.

closed 2022-03-04 by dkehlet