chipsalliance/aib-phy-hardware #36: v2 slave (follower) CSR differences

dkehlet, 2020-07-08:
h208 rx fifo
27  0 fifo single width
    1 fifo double width
0   0 no swap
    1 rxswap
    
h210 rx fifo
2:1 00 FIFO 1x
    01 FIFO 2x
    11 Register
0   0 word alignment disable, for FIFO 1x mode
    1 word alignment enable, for FIFO 2x mode

h218 tx fifo
23  0 disable word marking
    1 enable word marking
22:21 00 FIFO 1x
      01 FIFO 2x
      11 Register mode
0   0 no swap
    1 swap data from MAC to far side

Examples:
2x FIFO Mode
208	0A780204
210	02871F03
218	47A08004

Register mode
208	02780204
210	02871f06
218	47608004

weizhuintel, 2020-07-08:
Yes. V2 slave register definition is not the same as the master.
Please refer to section 5.6 of the V2 slave IP mas for all the detail description.

dkehlet, 2020-07-08:
Add a tab to the open_source_aib_csr.xlsx spreadsheet in the docs folder, and put the follower differences there.

weizhuintel, 2020-07-09:
Added to the spreadsheet. File sent to Julie. Julie will pull the file in.

closed 2020-07-09 by weizhuintel