chipsalliance/aib-phy-hardware #48: wait_xfer_ready doesn't return for rev1/ndsimslv

mkhazraee, 2020-10-05:
Hi,

I'm trying to run the ndsimslv testbench, but the o_ehip_init_status become 3'b100 and not 3'b111. I checked and all the signals from s10_slave interface are in done/out_of_reset state (config_done, fs_mac_ready, ns_mac_ready, ns_adapter_rstn, iopad_fs_mac_rdy, iopad_ns_mac_rdy). I also tried both REGISTER_MOD and without it. I went down couple modules and rx_rst_sm_cs transitions 2 states forward but stops (gets to WAIT_REMOTE_RX_DLL_LOCK but doesn't go to WAIT_REMOTE_RX_ALIGN_DONE). Am I missing some other setting? Is there some `force term that I should uncomment? or some cfg_avmm_write that I should uncomment?

I'm actually using cocotb+icarus to run the benchmark, it needed small modification to some logic/wire terms to make it work with the system verilog codes. But I doubt if that's causing any problems, as other signals are working properly and the warnings I get are totally fine. Can you run the modelsim (that's the code that I built the testbench upon) and see if it goes through? If so can you put an example waveform (or a temporarily link to one) for me to compare?

Thanks,
Moein

dkehlet, 2020-10-05:
Hello Moein,

Would you please apply the workaround described in issue #31?  Let me know if that has/does not have an effect.

Dave

xinyang2k, 2020-10-05:
Hello Moein,

I just tried modelsim with the ./runvsim command.  The issue Dave described might happened in the model that connect to the AIB channel. However, if you run the ./runvsim command directly, the default vlog.do use FPGA model. Without any change, you should be able to run.

The following is the modelsim version I ran today:

# 2020.2

# do self_test.do
# ../aib_lib
# ../rtl
# QuestaSim-64 vlog 2020.2 Compiler 2020.04 Apr 19 2020
# ** Warning: (vlog-13288) Multiple macros defined in +define+ command line switch.
# Start time: 10:02:56 on Oct 05,2020

Thanks,
Julie

mkhazraee, 2020-10-05:
Hi Dave, 

Their race condition is in aib_io files in /rev1/rtl but actually the original ndsimslv is not using those files. I also tried rev2 files for the same experiment, but same results. That being said, there is a icarus warning that now worries me:
s10aib/rtl/an.v: timing checks are not supported and delayed signal "delayed_clk_in_1" will not be driven. 

and it is given for the other clocks too. I thought that module is to deal with DDR, could it be causing this issue? 

Moein

mkhazraee, 2020-10-05:
Hi Julie, 

Just saw your reply. I was actually following the same file to replicate for cocotb, at the end of the file it also loads modules from "../maib_rtl/". I'll see maybe it doesnt' actually use them in modelsim because of a flag or something. But then that model would be less accurate on the internal state and time to calibrate, or they are practically the same?

Thanks,
Moein

xinyang2k, 2020-10-05:
Hi Moein,

The maib/rtl is FPGA main AIB. It is a model that is accurate if your AIB want to talk to FPGA.  You can check the user guide under doc for what it does.  It is more accurate model that include the actual DLL/DCC calibration.

Thanks,
Julie

mkhazraee, 2020-10-05:
Hi Julie, 

Yes, I want to eventually use s10 FPGA and first test the maib/rtl code. The part which is not clear is your do self_test.do only shows  ../aib_lib and ../rtl, but it should be using maib/rtl if I understood you correctly. 

Would you please do me a favor and share a link to the generated waveform so I can compare? I have to get and install vsim which I'm still waiting for intel people and gonna contact Mentor Graphics for an academic license.

Moein

mkhazraee, 2020-10-05:
Hi Dave and Julie, 

Actually it worked, just I had to let it run for a really really long time. Sorry for opening the issue. I'm gonna now try sending and receiving some data. Let me know if you are interested in what changes I made to make it compatible with icarus support of system verilog.

Moein

mkhazraee, 2020-10-05:
Actually let me ask, should it take about 11ms for wait_xfer_ready to be asserted, or something is not right? The signals look more random that eventually the load and raised signal happen to occur at the same time. (rx_aysnc_fabric_hssi_ssr_load happens one cycle after sr_fabric_rx_dll_lock_int until 11ms that sr_fabric_rx_dll_lock_int is high for a longer time. These are in u_c3aibadapt_wrap/c3aibadapt/adapt_rxchnl/rx_async/adapt_rxasync_update).

Moein

dkehlet, 2020-10-05:
Hi Mein, if you are willing to share what you did as a comment to issue #48, that would be welcomed!

Dave

mkhazraee, 2020-10-05:
For sure, I'll make a patch and will post it as a comment later today. 

One more problem, when I turned off the register mode, to get 78bit values on the FPGA side, o_echip_init_status doesn't get 3'b111, stops at 3'b110 even after 37ms. Also there is no data transmitted between the two sides. In register mode it was working properly, at least the data was being sent. I only changed the force values like nda_drv in ndsimslv, and the previous run was using the clocking for non-registered mode. Is there anything else I should change? 

Moein

xinyang2k, 2020-10-06:
I did not see any of the problem in the open source example if you follow up the REAME.txt in ndsimslv. It does not take ms long to run either.

Julie

mkhazraee, 2020-10-06:
Julie, thanks for the runtime information. I am following the readme and it's parameters, but I'm using a different simulator. Would you please upload a waveform so I can cross check? I am waiting for QuestaSim, but I don't get any bad warnings from icarus and it might be there is a typo somewhere in my copying. Also eventually I want to use icarus to be able to use python. 

I think there is a problem in serial data load. I assume I shouldn't uncomment any of the lines related to sr, such as this one:
yield cfg_avmm_write(dut, 0x314, 0xf, 0x00000000) # turn off sr_reserved_in_en/out_en sr_parity_en for spec compliance. 

It didn't help, but thought maybe there is some specific config for the sr module, as there are some commented in the nda_drv.sv too. 

Thanks,
Moein

mkhazraee, 2020-10-06:
Hi,

I still couldn't fix the problem, but my changes to the repo became more of a clean up process, so I'm sharing the patch. Here is what has changed in both revisions:

**Unsupported System Verilog changes:**
- Redundant "var logic" is changed to "logic"
- changed i+=1 in for loop to i=i+1 in /c3lib/rtl/lcell/c3lib_tie_bus_lcell.sv  
- c3lib/rtl/cdc/glitch_free_mux/c3lib_gf_clkmux.sv was modified slightly to use all 3 bits of clk_a_sel and clk_b_sel as wire, not 2 as wire and one as register. 

**Timescale clean up:**
- Timescales are now all 1ps/1ps. As this was the parameter in all the runs, and files that were not ps/ps were not using any delays. Actually some of the files already commented out timescale 1ns/1ns, so I just commented the rest. 
- c3aibadapt/rtl/c3aibadapt_avmm/c3aibadapt_avmm_cmdbuilder.sv used timestep of 100ps, I made it 1ps and changed the delays from 1 to 100
- aibcr3_lib/rtl/aibcr3_dcc_phasedet.v used the timestep of 100fs, and FF_DELAY is set to 200. But in every other file FF_DELAY is 200 with the timestep of 1ps. I assumed it should be 200ps, **let me know if that needs to be changed.** 

**Redundant warning removal:**
- There were very few files that had "begin begin * end end", I removed one pair to silent some of the warnings. I thought maybe it's for being nicer, but in the same file this is not consistent and not useful.   
[basic_sys_verilog_n_consistent_timescale_patch.zip](https://github.com/chipsalliance/aib-phy-hardware/files/5336593/basic_sys_verilog_n_consistent_timescale_patch.zip)

- There were very few files (5 I guess) that used the "unique case" while all the options were provided, even with redundant default. I just removed the word unique to be consistent with other files and silent icarus warnings.

If you can't upload the waveform and can email it to me, please send it to <my first name>@mit.edu 

Moein

mkhazraee, 2020-10-07:
Oh one more suspicious file I ran into and forgot to mention, in aibcr3_lib/rtl/aibcr3_dll_8ph_intp.v we have: 

`initial step = 2;  //min:1.5ps; typ:2ps; max:4ps
initial intrinsic = 50;  //min:10ps; typ:50ps; max:80ps

initial step = 10;  //min:1.5ps; typ:2ps; max:4ps 

always @ (*) ....

initial intrinsic = 50;  //min:10ps; typ:50ps; max:80ps`

Shouldn't the second step be also 2, or both repeated initials be removed? 

One other suggestion is to remove all // `timescale 1ns/1ns from the files. I wanted to be consistent, but I think since they are not used it's better to remove them, or maybe change it to 1ps/1ps as rest of the files.

mkhazraee, 2020-10-07:
Finally found the configuration which was messing things up (at least for me, soon I'll get modelsim and would double check):
`force `MAIB_DUT.`MAIB_SUB.aib_dprio_ctrl[39:0] = 40'h10000000 instead of 40'h18000000. `
This is setting ihssi_rb_dcc_byp_dprio to 0. 

The runtime is still long, but testbench finishes in 11us, that 11ms was the timescale problem. 

I've added my last comments to the patch, and used 20ps for aibcr3_lib/rtl/aibcr3_dcc_phasedet.v, assuming that 100fs was intentional. At the end of the day in rev2/rtl/v2_common/ana/ this delay is removed. As I said I removed the redundant initials from aibcr3_dll_8ph_intp.v, but again it is removed in rev2/rtl/v2_common/dig/bc_mod/ which makes it less important. 
[basic_sys_verilog_n_consistent_timescale_patch.zip](https://github.com/chipsalliance/aib-phy-hardware/files/5339622/basic_sys_verilog_n_consistent_timescale_patch.zip)

Please let me know if you have some insights on why ihssi_rb_dcc_byp_dprio being 1 is causing sl_tx_transfer_en not getting set. 

Moein

mkhazraee, 2020-11-04:
The bug was coming from rev2/rtl/v1_slave/s10aib/rtl/an.v where $setuphold is used and icarus doesn't support it properly. I replaced the delayed assignment with normal assign and it worked. This was causing issues in DCC, and hence bypassing it was a workaround. Thanks for your feedback, I'm closing the issue.

closed 2020-11-04 by mkhazraee