chipsalliance/aib-phy-hardware #59: AIB Adapter Word Marking and Word Assembly question

duytrinhkhac, 2022-02-28:
Hello,

In section 2.2.2.2, there are sections as below:
"Once aligned, the adapter does not need to locate the Mark bit again. This permits the transmit side to stop sending Mark bits and instead use that bit position for data.
Once aligned, if the adapter sees an unexpected 0 or 1 in the Mark bit position, the adapter shall set m_rx_align_done LO."

I have 2 questions:
- How RX side notifies TX side to stop sending Mark bit?
- When TX side stops sending Mark bit, how does RX monitor Mark bit and pull the signal m_rx_align_done to low?

Could you help me please?

dkehlet, 2022-03-04:
Hello duytrinhkhac,

	_How RX side notifies TX side to stop sending Mark bit?_
This is outside of the AIB spec. In practice we use a side channel such as SPI to set registers in the far side chiplet. An AIB implementation needs to have a control register "tx_wm_en" that enables or disables word markers in the TX words. For TX on the far side chiplet, the side channel (e.g. SPI) is used to send a write to the tx_wm_en bit. For TX on the near side, there needs to be a local register set with tx_wm_en that a controller can write to.

    _When TX side stops sending Mark bit, how does RX monitor Mark bit and pull the signal m_rx_align_done to low?_
The RX side needs to have Mark bit monitoring logic. This logic looks for the Mark bit, and if it is not seen where expected then the monitoring logic pulls m_rx_align_done low.

dkehlet, 2022-03-04:
See file aib-phy-hardware/v2.0/rev1/rtl/aib_channel.v and search for tx_wm_en.

duytrinhkhac, 2022-03-05:
Hell dkehlet,

Thanks for the answer, but what I meant in #2 was: Because TX stops sending mark bits once aligned (see the last sentence of the first quoted paragraph) , how RX still can monitor them?

duytrinhkhac, 2022-03-08:
Hello Dave,

I think I understand the idea now, could you please confirm if it is correct or not:

- TX sends Mark bits
- RX senses Mark bits and does alignment. Once aligned, m_rx_align_done = 1. The data alignment is stored.
- TX stops sending Mark bits and transfers data only.
- RX can't senses Mark bits, thus m_rx_align_done = 0. But the data alignment is still kept.
- If TX needs to resend Mark bits with different alignment, reset must be done beforehand.

dkehlet, 2022-03-08:
Yes, this use as described above is anticipated by the AIB spec.

duytrinhkhac, 2022-03-09:
Many thanks for your clarification.

closed 2022-03-09 by duytrinhkhac