chipsalliance/aib-phy-hardware #61: Gen2 Mode control in AIB2.0

clesnps, 2022-03-11:
I would like to confirm the function of signal m_gen2_mode in the v2.0 model since I cannot find it in the AIB 2.0 spec.  Does m_gen2_mode = 0 mean the model work in Gen1 mode in the spec and can compile with AIB 1.0 model? And m_gen2_mode = 1 means the model work in Gen2 mode in the spec and can compile with AIB 2.0 model?

dkehlet, 2022-03-11:
Hi, please see page 33 in the 2.0 spec:
[image]

m_gen2_mode is only present in the AIB 2.0 model located here:
[image]

clesnps, 2022-03-12:
Hi @dkehlet, can you give me the link to the 2.0 spec referred to? I cannot find that signal in the spec here :
[image]

[image]

dkehlet, 2022-03-13:
Hi, the AIB specification was moved about 18 months ago to:
http://github.com/chipsalliance/AIB-specification 
Sorry about the old version under aib-phy-hardware/docs. I will delete it ASAP.

clesnps, 2022-03-14:
Hi @dkehlet, thanks for your clarification.
So as my understanding, Gen1 mode of v2.0 model can compile with v1.0 model (AIB 1.0 spec), is that correct?

dkehlet, 2022-03-14:
Hi @clesnps v2.0 model in Gen1 mode can simulate and operate in hardware with v1.0. Please see the README in this directory for the simulation example:
https://github.com/chipsalliance/aib-phy-hardware/tree/master/v2.0/rev1/dv/sims

clesnps, 2022-03-15:
Thanks @dkehlet for your fast reply. It's clear for me now.

closed 2022-03-15 by clesnps