chipsalliance/aib-phy-hardware #62: Voltage level of *_RCV_CLK should be 0.9V or 0.4V

nguyentheman90, 2022-03-24:
Hi Team,

I have a question related to dual-rail voltage. From the Table 21 in the AIB 2.0 spec. I understood:
  + Gen 1 mode: all Bump work at 0.9V
  + Gen 2 mode: 
        > TX,RX,NS_FWD_CLK,NS_FWD_CLKB,FS_FWD_CLK, FS_FWD_CLKB run at 0.4V
        > Remain IOs run at 0.9V

[image]

However when redundancy is active:
+ the NS_RCV_CLK/NS_RCV_CLKB will be run as TX[0] and TX[1]
+ the FS_RCV_CLK/FS_RCV_CLKB will be run as RX[0] and RX[1]
=> So it mean the *_RCV_CLK should be work as 0.4V as TX/RX bump. Am I correct?

Regards!
Man Nguyen

daniwrk, 2022-03-25:
Hi Man Nguyen,

Thanks for the question. This is an excellent point.

This case is applicable to AIB Plus only since ns_rcv_clk/b only exists in AIB Plus.

Similar to other IOs such as Tx[19:0] and Rx[19:0] that are required to operate at 0.9v and 0.4v in Gen1 and Gen2, respectively, the IO’s connected to AIB43/ns_rcv_clkb and AIB42/ns_rcv_clk should also be able to operate at 0.9v and 0.4v as follows:

Gen1:   Redundancy Disable:      Signal at 0.9v as ns_rcv_clkb and ns_rcv_clk (AIB Plus only)
            Redundancy Enabled:     Signal at 0.9v as Tx[1] and Tx[0]

Gen2:  Redundancy Disable:       In standby mode (the receive clock is not used in Gen2 mode, see 2.1.6.2 “Receive Clock (AIB Plus only, Gen1 mode only”)
           Redundancy Enabled:     Signal at 0.4v as Tx[1] and Tx[0]

(same is true for AIB59/fs_rcv_clk, AIB58/fs_rcv_clkb).

Also, please see section 3.3.1.1. “Neighboring-Cell Requirements”.

Dani

dkehlet, 2022-04-04:
Hello nguyentheman90, is this issue closed?

nguyentheman90, 2022-04-04:
Hi daniwrk and dkehlet

So sorry, I forgot to put the comment. The issue can be closed now.  It is cleared for me

Regards!
Man Nguyen