chipsalliance/aib-phy-hardware #64: AIB Adapter Word Marking and Word Assembly questions (part II)

tramphamsnps, 2022-04-04:
Hi Team,

To be continue the confirmation about word marking/assembly flow at https://github.com/chipsalliance/aib-phy-hardware/issues/59, I'd like to confirm more my understand about word marking/assembly:
1. There are 2 options:
-PHY executes word marking/assembly internally.
=> m_rx_align_done is used to inform to MAC know that data alignment is done.
-MAC executes word marking/assembly by itself. 
=> m_rx_align_done is not used. it's because Tx MAC send marking bit through data_in_f
and Rx MAC receive marking bit through data_out_f and MAC process the alignment by itself.

2. When Tx sends marking bits, other bits shall be zero and are unavailable during this process because data alignment has not done 

Thank you in advance for your support.

Best regards,
Tram Pham

dkehlet, 2022-04-04:
1. Yes.
2. Tx only overrides the Mark bit locations in data_in_f. Other bit locations are not changed by the word marking.

tramphamsnps, 2022-04-04:
Hi Dave,

Thanks for your quick confirmation
2. It means that MAC can use other bit locations to transfer data for the purpose of MAC during word marking/assembly handshaking, doesn't it?
I used to understand that other bit locations are unavailable data (dummy data) because data alignment has not done

Best regards,
Tram Pham

dkehlet, 2022-06-15:
_2. It means that MAC can use other bit locations to transfer data for the purpose of MAC during word marking/assembly handshaking, doesn't it?_

Hi, this might be possible but as the alignment is achieved there might be loss of data words as it moves from unaligned to aligned. The exact behavior is unspecified.

closed 2022-07-27 by dkehlet