chipsalliance/aib-phy-hardware #66: sim_dcc and sim_dcc_static

CapucinedeBoissac, 2022-07-27:
Hello, 

I have been running the DCC simulation in v1.0/rev1/how2use/sim_dcc and I do not understand what I am supposed to observe; the README.md promises these two to show how DCC works and correct the 40/60 input duty cycle to a quasi-50/50. Given the test files there should also be a loopback test following the calibration of the DCC, but in fact no xx_dcc_cal_done is set and clk_dcc does not behave as a 50/50 clock at all (see screenshot).

Any hint on what I am missing?

Many thanks in advance.

Capucine
[image]

xinyang2k, 2022-07-27:
Hi Capucine,
If you want to look at the internal signal regarding correction of 40/60 duty cycle distortion, you should look at the following path:
top.dut.u_c3aibadapt_wrap.xaibcr3_top_wrp.xaibcr3_top.xrxdatapath_rx.x1591.dcc_done .  The xrxdatapath is not a typo, due to legacy reason, the name convention is that the transmit path called rxdatapath (Because at the time, rx mean  from the FPGA point of view. So rx is actually tx and tx is actually rx. So in your Rx DCC, it is actually tx DCC.)
I re-run at the my side with vcs simulator, this is what I got:
[image]

After zoom in, you can see the output clk_dcc is almost 50/50 while the i_rx_pma_clk and clk_dcd is 60/40:
[image]

With different simulator, I do see they may behave a little bit different, but the final result should be correct.

--Julie Zhang

CapucinedeBoissac, 2022-07-28:
Many thanks for the hint. Turns out Questasim does not allow inout ports to be directly assigned values. (c.f. iopad_fs_sr_data and iopad_fs_adapter_rstn in local top.sv).
Is there a more detailed description on how the DCC, and the FSM that drives the DCC calibration work ?

Regards.

xinyang2k, 2022-07-28:
You can look at https://github.com/chipsalliance/aib-phy-hardware/blob/master/docs/FCCM_2019_Workshop_AIB.pdf  start with page 31 to get more information.  
The Questasim example has been supplied here that you can reference:https://github.com/chipsalliance/aib-phy-hardware/tree/master/v1.0/rev1/ndsimslv

CapucinedeBoissac, 2022-08-02:
Many thanks Julie for your help.

Capucine

xinyang2k, 2022-08-02:
You are welcome. The issue will be closed.

closed 2022-08-02 by xinyang2k