chipsalliance/aib-phy-hardware #68: Repair of TX0 pin does not work v2.0/rev1/rtl/aib_ioring.v

tramphamsnps, 2022-08-22:
Hi,

When do repair from Tx0 to ns_rcv_clk, Tx0 is not shifted down to ns_rcv_clk.
I think the cause is from the connection of launch clock of io. It should be from tx_ilaunch_clk  instead of ns_rcv_clk_frmac
[image]

This issue was reported at #44 , it is already fixed in v1.0/rev1/rtl/aib_ioring.v but it happens again in v2.0
[image]

Could you check this issue?

Thanks,

dkehlet, 2022-08-22:
Hello, can you specify which RTL code you are referring to?
rev1? rev1.1?

dkehlet, 2022-08-22:
Sorry, just saw the issue title (rev1).

dkehlet, 2022-08-24:
Hello Tram,

Can you please try the v2.0/rev1.1 code? It is a separate implementation that is going in to silicon.

Meanwhile, if you wish to propose changes to rev1 to correct this and the other issue, please submit a pull request. We will review ASAP.

tramphamsnps, 2022-08-29:
Hi Dkehlet,

Our scope is for rev1 only, we expect rev1 need to be updated.
Because I'm not familiar to github and the update is not much, please help to update rev1 based on the description "**Recommend fixing ...**"  in this issue report #68 and #69 

Thanks,
Tram Pham

closed 2022-08-29 by xinyang2k