chipsalliance/aib-phy-hardware #72: DCC and DLL verilog code explanation

CapucinedeBoissac, 2022-09-08:
Hello, 

Do you have further information (slides, article...) to provide to explain DCC, DLL FSMs, structure and functioning? I find the comments and the documentation (2 slides in FCCM Workshop 2019) rather imprecise.

Many thanks in advance for you help.

Capucine

xinyang2k, 2022-09-15:
Hi Capucine,

Could you be specific about what are you trying to do and need? 

Thanks,
Julie

CapucinedeBoissac, 2022-09-21:
Hello Julie, 

Apologies for the unclear request. I was wondering if this DLL/DCC hybrid module is inspired from litterature, or if this was a completely custom IP. 

That being said, I have several questions regarding the module functioning, some might overlap: 

1- I backtraced the _clk_dcd_ signal in [...]aibcr3_top/xrxdatapath_rx/dcc_top and the original source is _m_ns_fwd_clk_ from top, which surprised me since I thought the DCC would be on the receiver side; I mean that I was expecting the source clock to be coming from the receiver I/O pads ( _iopad_fs_fwd_clk(b)_ ) and not the forward clock from the MAC. 
2- Where do xxx_clk and xxx_clkb merge, and where do they split?
3- I have trouble forward-tracing the output _clk_dcc_ of dcc_top: I can see it entering xclktree but it is quite difficult to trace it further; some outputs even seem to end up in dead-end paths.
4- What is the function of the clktree and the clktree replica? Plus the _clktree_out_ input of dcc_top seems to have a static '0' input.
 In a nutshell: I can see the sim_dcc[...] simulations work but I hardly understand through back and forward tracing the relationship between the differential transmit clocks and the single clock being corrected by the DCC.

Many thanks for you help.

Capucine

CapucinedeBoissac, 2022-10-25:
Hello Julie, I come back to you with questions now concerning the V2.0/rev1.1 RTL.

I can clearly identify the specs’ datapath calibration flow in the adapter RTL. But I am surprised not to find any signal being sent to any FSM/analog model during the states SEND_[…]_REQ. However, there are input signals labelled xx_dcc_cal_done or xx_dll_lock that enable the calibration flow to continue. Those two input signals are both coming from the same FSM rcomp_calibration_fsm which I understand is neither a DLL/DCC calibration FSM, nor connected to one. In the meantime, I can observe that a lot of (FSM, model) couples are involved in the link init./training and whose names are a mystery. In short, I fail to understand where the DCC and DLL are, and when their calibration is supposed to occur.
-	Can you please clarify what are the functions of dcs_fsm, cdr_fsm, phase_adjust_fsm, ntl_fsm, and the building blocks attached to them?
-	What are tx_dll and i_xx_deskew for?

Many thanks in advance for your help.

Capucine

xinyang2k, 2022-10-25:
Hi Capucine,

I assume you are referring https://github.com/chipsalliance/aib-phy-hardware/tree/master/v2.0/rev1.1/rtl/bca
This logic is an implementation from third party of AIB hard macro.  The project is still ongoing and we will release documentation later. I will forward your comment to them.  Due to the resource limitation and tape out pressure, the response time will be delayed.

Julie

CapucinedeBoissac, 2022-10-27:
Hello Julie, yes I was referring this RTL code. I look forward to reading your comments on this. Thanks for your help, courage and good luck for the tapeout.

Capucine

xinyang2k, 2024-07-29:
Hi Capucine,

The functional specification has been uploaded here: https://github.com/chipsalliance/aib-phy-hardware/blob/master/docs/BCA_AIB_Gen2_PHY_Functional_Spec_v1p23_github.pdf

Thanks,
Julie

closed 2024-07-29 by xinyang2k