chipsalliance/aib-phy-hardware #73: Clarify file hierarchy and versions vs. spec revisions

CapucinedeBoissac, 2022-09-26:
Hello, 

Can you please clarify the relationships between the specification revisions and the AIB generations? 

- v1.0/rev1 contains the RTL describing the specs revision 1.2, i.e. the AIB Gen1 (Plus); but then why does how2use contain a simulation involving the phase compensation fifo, which is not supposed to be implemented in v1.0?
- V1.0/rev2 contains the RTL corresponding to the AIB Gen2; Then why is it the v1.0/rev1 code which was updated more recently?
- What are v2.0/rev1 and rev1.1?

Thanks in advance for your help

Capucine

dkehlet, 2022-09-26:
Hello Capucine, 
_v1.0/rev1 contains the RTL describing the specs revision 1.2, i.e. the AIB Gen1 (Plus); but then why does how2use contain a simulation involving the phase compensation fifo, which is not supposed to be implemented in v1.0?_
The v1.0 code contains the FIFO which is not required by the AIB 1.x spec. It is extra functionality we provided. 

_V1.0/rev2 contains the RTL corresponding to the AIB Gen2; Then why is it the v1.0/rev1 code which was updated more recently?_
That directory had old files that were cleaned up. I recommend you to use v1.0/rev2 if you want to implement AIB 1.0 (spec revision 1.2).

_What are v2.0/rev1 and rev1.1?_
v2.0/rev1 is a behavioral model of AIB 2.0. rev1.1 is RTL extracted from an actual AIB 2.0 design. Functionally these are intended to be equivalent. rev1 simulates a lot faster than rev1.1, so for a lot of regressions people prefer to use rev1. For a tapeout, a user should run rev1.1 for final verification.

Dave

dkehlet, 2022-09-27:
_what are the main differences between v1.0 and v2.0?
what do I earn by working with 2.0 RTL?_
From page 20 of xxx:
[image]

In short, AIB 2.0 gives 4x the bandwidth in the same die shoreline as AIB 1.0.
That said, if you want to interface with a Stratix 10 or Agilex FPGA (and you would need a sponsor and packaging agreement with Intel for this), then AIB 1.0 is sufficient.

closed 2023-09-01 by dkehlet